esp-rs/esp-pacs #57: MCPWM TIMERx_SYNC is incorrect for ESP32-S3

dimpolo, 2022-11-24:
Correct definition for the ESP32
```xml
          <name>TIMER0_SYNC</name>
          <addressOffset>0xC</addressOffset>
          <size>0x20</size>
          <fields>
            <field>
              <name>TIMER0_SYNCI_EN</name>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>SW</name>
              <bitOffset>1</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>TIMER0_SYNCO_SEL</name>
              <bitOffset>2</bitOffset>
              <bitWidth>2</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>TIMER0_PHASE</name>
              <bitOffset>4</bitOffset>
              <bitWidth>16</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>TIMER0_PHASE_DIRECTION</name>
              <bitOffset>20</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
          </fields>
```

Missing `TIMER0_PHASE_DIRECTION` and incorrect `bitWidth` for `TIMER0_PHASE` for the ESP32-S3
```xml
          <name>TIMER0_SYNC</name>
          <description>PWM timer0 sync function configuration register.</description>
          <addressOffset>0xC</addressOffset>
          <size>0x20</size>
          <fields>
            <field>
              <name>TIMER0_SYNCI_EN</name>
              <description>When set, timer reloading with phase on sync input event is enabled.</description>
              <bitOffset>0</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>SW</name>
              <description>Toggling this bit will trigger a software sync.</description>
              <bitOffset>1</bitOffset>
              <bitWidth>1</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>TIMER0_SYNCO_SEL</name>
              <description>PWM timer0 sync_out selection, 0: synci, 1: TEZ, 2: TEP, otherwise:sync out is software sync</description>
              <bitOffset>2</bitOffset>
              <bitWidth>2</bitWidth>
              <access>read-write</access>
            </field>
            <field>
              <name>TIMER0_PHASE</name>
              <description>phase for timer reload on sync event</description>
              <bitOffset>4</bitOffset>
              <bitWidth>17</bitWidth>
              <access>read-write</access>
            </field>
          </fields>
```

jessebraham, 2022-11-25:
Thanks for letting me know, I'm off work today but I have some other patches which need to be added so I will take care of this on Monday with those and publish new PACs.

dimpolo, 2022-11-28:
Damn, sorry I was unclear :/
TIMER1_SYNC and TIMER2_SYNC actually also need patches

dimpolo, 2022-11-28:
All good!
(maybe you want to reopen this issue until then)

PR #58 (2022-11-28): Apply patches to `esp32` and `esp32s3` packages
- `SHA` registers/fields for ESP32
- `MCPWM` registers/fields for ESP32-S3
- Bump version numbers for both packages

Closes #57
PR #59 (2022-11-29): Additional corrections for ESP32-S3's MCPWM peripheral
Okay, I think this should be correct this time 😁 @dimpolo are you able to confirm nothing else needs patching at this time for this peripheral please?

(Hopefully) closes #57
closed 2022-11-29 by jessebraham