esp-rs/esp-pacs #154: Typo in SVD files

bugadani, 2023-08-22:
My bet is that this should be `QUERY` instead of `QUREY`. HMAC seems to consistently misspell it, though, so I'm not sure, but I think it doesn't hurt to have an issue to track it.

https://github.com/esp-rs/esp-pacs/blob/dd1bf1eefbc86b149053d32b1a1a41e39e59025e/esp32s2/svd/esp32s2.base.svd#L7156

jessebraham, 2023-08-22:
Hmm, that's kind of a funny one. The TRM can't even seem to agree on which to use 😁 

[image]

I assume it's a typo as well, I will patch any occurrences of this away at some point (or somebody else is welcome to if they really want to!)

bugadani, 2023-08-22:
lovely, but I'm not sure it's worth the time until it's actually used in esp-hal

jessebraham, 2023-10-26:
Fixed in the latest SVD

closed 2023-10-26 by jessebraham