esp-rs/esp-pacs #183: esp32s3 (maybe others) missing dedicated gpio?

yanshay, 2024-01-01:
I can't find the DEDICATED_GPIO in esp32s3 while in espressif docs this functionality is avaliable in that model. 

It is available in esp32s2.

Is it missing for a reason?

jessebraham, 2024-01-02:
Maybe I'm missing something, but as far as I can tell there are no actual registers for this peripheral? The header file linked in the documentation you shared isn't even in the `soc` component of ESP-IDF, which has been the case for all other peripherals as far as I can recall.

Is there something specific missing that you're looking for?

yanshay, 2024-01-02:
After searching some more, it seems like they provide that functionality but not using registers but rather through specialized cpu instructions:
https://www.esp32.com/viewtopic.php?t=27857

That probably means that `esp-pacs` is not the place to expect that to be, right?

jessebraham, 2024-01-02:
Yeah from the sounds of that we will need some driver code in another package somewhere, whether that's `esp-hal` or somewhere else. The scope of the PACs is really just to provide register access and enumerate the peripheral interrupts, so not really anything to do here.

Thanks for opening the issue regardless, though!

closed 2024-01-02 by jessebraham