esp-rs/esp-pacs #306: Moving SVD patching to the SVD repo

antoniovazquezblanco, 2024-12-31:
I find quite confusing the fact that up to date SVD files are in this repo and not in https://github.com/espressif/svd.

Would it make sense to move SVD patches over to the https://github.com/espressif/svd repo?

I think that would have the following benefits:
* Single source of SVD files
* SVD issue tracking will be handled in a single source of truth instead of being split into two repos
* esp-pacs issues would not be tainted with svd-related issues/PRs
* esp-pacs workflow would benefit from pulling changes from a repo instead of being generated from changes that would later be updated in another repo to be brought back...
* It would be easier for contributors that may want to provide SVD patches. Also easier for reviewers...

Any thoughts on this?
Would you accept PRs regarding this?
Thanks

antoniovazquezblanco, 2024-12-31:
I've just realized that this org may not be part of esp...

Sorry for the issue if that was the case. I just would love to help on this as much as I can if possible.

jessebraham, 2025-01-02:
I was going to write a long-winded reply to this, but I spoke with my team lead today regarding this, and we are working on coming up with a game plan moving forward. So, please give me a week or two, and I will let you know what we come up with :)

Apologies for letting things fall out of sync as much as they have, but unfortunately the work on SVDs up until now has been squarely on my shoulders, and I've been quite busy with more pressing matters 😅

antoniovazquezblanco, 2025-01-02:
Thanks for keeping track of this and for hearing us out. Also, thanks for getting back. Unfortunatelly it is not that common to see this kind of responses among many big comanies and that speaks out about the work you do.

I am sorry to hear that this has been only on you. The intention of this issue is to try and improve things with the least effort to you. I am sure others also have the same intent so please, let us know how shall we proceed to make your life easier.

Thank you!

antoniovazquezblanco, 2025-01-20:
Hi @jessebraham, any updates on this? Any way we may help?

Thanks!

jessebraham, 2025-01-20:
Sorry, still haven't heard anything back regarding this. @MabezDev any updates here?

MabezDev, 2025-01-21:
Thanks for the suggestion @antoniovazquezblanco! Unfortunately we make various naming changes among other things which means the SVDs here aren't suitable for `espressif/svd`. You can of course continue to use these provided you adapt to the naming conventions, and patches to these SVDs are always welcome. Our team no longer maintains `espressif/svd`, but I believe there are plans to keep that repo updated with new chips in the future.

closed 2025-01-21 by MabezDev

antoniovazquezblanco, 2025-01-21:
I am sorry to hear that. As I said before, from the outside it is sometimes difficult to understand where are the latest SVD files... It also makes contributing harder... Thanks anyways!

jessebraham, 2025-01-22:
> I am sorry to hear that. As I said before, from the outside it is sometimes difficult to understand where are the latest SVD files... It also makes contributing harder... Thanks anyways!

Yeah, I can sympathize and agree with you entirely. Unfortunately not a great situation, sorry there wasn't a better outcome here.