esp-rs/esp-pacs #331: Change NRX offset to `3ff5c000`?

Frostie314159, 2025-05-10:
While reverse engineering Wi-Fi CSI, I found that the NRX peripheral offset may be wrong. I know, that in [the blogpost](https://developer.espressif.com/blog/esp32-memory-map-101/) the address offsets are different, but for the S2 it starts at `3f41c000`. Configuring CSI requires five registers, starting from `xxxx_c000`:
1. `074` - `???`
2. `c0c` - `BB_CHAN_FILTER` (These are preliminary names, inferred through RE)
3. `d04` - `???`
4. `d08` - `???`
5. `d0c` - `BB_CHAN_DUMP`
Their function fits with NRX's purpose of controlling the RX path inside the baseband.  Therefore, I suggest changing the offset to the proposed value.

bugadani, 2025-07-17:
`3ff5c000` is not an address accessible via either PeriBus1 or PeriBus2.

Frostie314159, 2025-07-17:
My original message was incorrect. That address is only correct for the ESP32. For the S2 it will be different.

Frostie314159, 2025-07-19:
I definitely understand the argument. My explanation for why it would be different in `esp-idf` is, since AFAIK the only public register in the whole block is `NRXPD_CTRL`, because all other registers aren't relevant outside the blob. 

Also, keeping NRX at this address would mean, that it's the only peripheral address space, that doesn't start at an address aligned to 4096 bytes. All others start at `3ffx_0000` and are a multiple of 4kB long. 

This is only an issue for the register at `3ff5c074`, since it's outside any peripheral's address space.